Romkabouter/ESP32-Rhasspy-Satellite #133: Compatibility with Home Assistant Voice (Whisper / Piper)

chiefymuc, 2023-10-15:
I'd like to ask if there is a way to get this cool project to talk to the recently introduced Home Assistant Voice Features with mostly Whisper (and potentially Piper)? It looks like the Rhasspy developer is working on this architecture now, so I'd like to migrate as well.

Reason for my question is that I have a Matrix Voice with ESP32, and it seems that due to the FPGA/Microphone Architecture, esphome (which is to my understanding the main method to get ESP32s with microphone to talk to HA) would not support this for now (not a programmer, so this a bit beyond my ability of understanding).

Anyways, thank you for the amazing work on this project!

Romkabouter, 2023-10-18:
Should be possible yes! 
But I am currently very busy with other projects, so time is an issue here

markusressel, 2023-10-24:
If this was implemented, I think it could revive a lot of dead hardware :open_mouth: . My Matrix Voice is only catching dust right now, integrating it with Home Assistant would be huge! I could finally start to use my Matrix Voice after all this time of it sitting in a dark corner and catching dust :cry: 

I know very well how hard it is to find time for projects, I am struggling daily with that issue myself. If there is anything I can do to help, please let me know!

andersonsmith, 2023-10-25:
I am in the same boat with my matrix voice.  I looked at creating a custom microphone component within esphome but it looks as though only i2s microphones are in there at this time and no option exists for creating a custom one yet.

Romkabouter, 2023-11-06:
> My Matrix Voice is only catching dust right now, integrating it with Home Assistant would be huge! I could finally start to use my Matrix Voice after all this time of it sitting in a dark corner and catching dust

Well you can integrate with Home Assistant already!
Check the WIKI pages here (a large part of it written by me)
https://github.com/rhasspy/rhasspy/wiki

I know that is for 2.5, but will probably work with 3 as well.

WhiteSockedDancer, 2023-11-16:
I'm also using rhasspy 2.5 with your esp32-Rhasspy-Satellite Streamer on my Matrixvoice. 
I also think about how to use the device with Rhasspy 3 (probably also integrated into Homeassistant) 

I found another Project that seems to try the same for Matrixvoice Devices by integrating it into esphome:
https://github.com/gnumpi/esphome_matrixio
I didnt try that yet since i'm still running rhasspy 2.5 and i dont know how to flash OTA and i don't want to get my old RPi running again.