Romkabouter/ESP32-Rhasspy-Satellite #134: Distorted mic signal with Inmp441Max98357a device

requireiot, 2023-11-28:
With the Inmp441Max98357a device, the audio input (microphone signal) is distorted, because each sample is _sent twice_ (apparently, this is what the ESP32 API will do, when 16 bit resolution is requested).

To illustrate, I recorded the streamed signal while playing a loud 440 Hz tone through a PC speaker, the read the resulting WAV file in Audacity. Below are screenshots from Audacity, from the zoomed WAV signal, amplified by 6 dB for clarity

With the original code
[image]

With modified code, using 32 bit samples
[image]

The modified code is in my [fork](https://github.com/requireiot/ESP32-Rhasspy-Satellite) of the original code. I'm not a Github expert, I guess I'll have to figure out how to do a pull request ...

Romkabouter, 2023-12-09:
Nice fix, I will check your fork.
Thank you for the effort!

requireiot, 2023-12-09:
The code in my fork has 2 changes compared to the original, which maybe I shouldn't have checked in together, but too late now:
1. a _functional_ change to implement the duplicated 16-bit words as described above
2. a _cosmetic_ change, making the definitions at the top of `Inmp441Max98357a.hpp` more consistent: all pin numbers etc. pertaining to the microphone I2S port start with MIC_, all pin numbers etc. pertaining to the speaker I2S port start with SPK_, and the config structs in `Inmp441Max98357a::init()` end with _mic or _spk, resp. -- just made it easier for me to wrap my head about what pertains to what