Romkabouter/ESP32-Rhasspy-Satellite #136: Matrix Voice 1 panic'ed (IntegerDivideByZero)

jlaba, 2023-12-16:
Following the Readme I uploaded the firmware to my matrix voice. Connected with minicom to /dev/ttyS0 it shows a reboot loop with the following output:

`LF file SHA256: 0000000000000000

Rebooting...
I (10) boot: ESP-IDF v3.1 2nd stage bootloader
I (10) boot: compile time 11:08:24
I (10) boot: Enabling RNG early entropy source...
I (14) boot: SPI Speed      : 40MHz
I (18) boot: SPI Mode       : DIO
I (22) boot: SPI Flash Size : 4MB
I (26) boot: Partition Table:
I (29) boot: ## Label            Usage          Type ST Offset   Length
I (37) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (44) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (52) boot:  2 app0             OTA app          00 10 00010000 001e0000
I (59) boot:  3 app1             OTA app          00 11 001f0000 001e0000
I (67) boot:  4 eeprom           Unknown data     01 99 003f0000 00001000
I (74) boot:  5 spiffs           Unknown data     01 82 003f1000 0000f000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x38800 (231424) map
I (176) esp_image: segment 1: paddr=0x00048828 vaddr=0x3ffbdb60 size=0x05a58 ( 23128) load
I (185) esp_image: segment 2: paddr=0x0004e288 vaddr=0x40080000 size=0x01d90 (  7568) load
I (188) esp_image: segment 3: paddr=0x00050020 vaddr=0x400d0020 size=0xce040 (843840) map
I (489) esp_image: segment 4: paddr=0x0011e068 vaddr=0x40081d90 size=0x13234 ( 78388) load
I (522) esp_image: segment 5: paddr=0x001312a4 vaddr=0x50000200 size=0x00010 (    16) load
I (535) boot: Loaded app from partition at offset 0x10000
I (535) boot: Disabling RNG early entropy source...
Booting
Matrix Voice Initialized
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dac29  PS      : 0x00060330  A0      : 0x800dac94  A1      : 0x3ffb26b0  
A2      : 0x3ffbb9e4  A3      : 0x3ffbb9e4  A4      : 0x00000000  A5      : 0xffffffbf  
A6      : 0xffffffdf  A7      : 0x3ffb9374  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ffb26bc  A13     : 0x00000004  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40089ee9  LEND    : 0x40089f0b  LCOUNT  : 0xffffffff  

Backtrace:0x400dac26:0x3ffb26b00x400dac91:0x3ffb26e0 0x400d305f:0x3ffb2760 0x400d8062:0x3ffb27a0 0x400ef56e:0x3ffb2820`

This is my settings.ini:
`[General]
hostname=192.168.2.195
deployhost=192.168.2.103
siteId=satellite
;supported: M5ATOMECHO=0, MATRIXVOICE=1, AUDIOKIT=2, INMP441=3, INMP441MAX98357A=4, ESP32_POE_ISO=5, TAUDIO=6, INMP441MAX98357AFASTLED=7
device_type=1
;network_type: 0: WiFi, 1: Ethernet
network_type=0

[Wifi]
ssid=MY_SSID
;ssid=MY_PASSWORD
;password=Gothenburg2021
password=DidWdHH15HB
;scanStrongestAP=true ; use if device connects to a non-optimal AP

;uncomment next 4 lines and fill with your own values if you want to use a static ip address
;ip=192.168.xxx.xxx
;gateway=192.168.xxx.xxx
;subnet=255.255.255.0
;dns1=192.168.xxx.xxx
;optional: second dns server
;dns2=192.168.xxx.xxx

[OTA]
;supported: upload, ota, matrix
;-upload: device should be attached to computer via usb
;-ota: will use espota
;-matrix: matrix voice should be attached to a raspberry pi with matrix software.
;         deployhost should be set to ip of the pi
method=ota
password=DiesPfO\!2022
port=3232

[MQTT]
hostname=MY_IP
port=1883
username=MY_USERNAME
password=MY_PASSWORD`

To setup the environment on the PI I have used this repo https://github.com/qnlbnsl/Matrix-IO since the matrix.io repos are offline. (See separate issue to add a hint to your Readme file for that).

Romkabouter, 2023-12-16:
Is the hostname in the MQTT section your IP or the value MY_IP?
Don't know if that is the issue by the way.
And maybe edit you passwords out ;)

jlaba, 2023-12-16:
Good tip :-) (passwords)

The hostnae is my mqtt ip address.