Romkabouter/ESP32-Rhasspy-Satellite #138: Developpent of new esp

orrionis, 2024-01-19:
Hello,
i purchase this

[https://www.espressif.com/en/news/ESP32-Korvo](url)

for technical information:

[https://github.com/espressif/esp-skainet/blob/master/docs/en/hw-reference/esp32/user-guide-esp32-korvo-v1.1.md](url)

I want to use with rhasspy, i have a just an esphome code:

`esphome:
name: ${device_name}
min_version: 2023.11.0
platformio_options:
board_build.flash_mode: dio
on_boot:
- priority: -100
then:
- wait_until: api.connected
- delay: 1s
- if:
condition:
switch.is_on: use_wake_word
then:
- voice_assistant.start_continuous:

esp32:
board: esp-wrover-kit
framework:
type: arduino
version: recommended

external_components:

source: github://rpatel3001/esphome@es8311
components: [ es8311 ]
source: github://rpatel3001/esphome@es7210
components: [ es7210 ]
source: github://pr#5230
components:
esp_adf
Enable logging
logger:

API KEY
api:
encryption:
key: ${apikey}

OTA PASSWORD
ota:
password: ${wifi_password}
safe_mode: true
reboot_timeout: 3min
num_attempts: 3

text_sensor:

platform: wifi_info
ip_address:
name: “${device_name} IP Address”
time:
platform: homeassistant
id: homeassistant_time

WIFI PASSWORD
wifi:
networks:
- id: wifi_default
ssid: ${wifi_ssid}
password: ${wifi_password}

ap:
ssid: ${device_name}
password: ${wifi_password}

i2c:
sda: GPIO19
scl: GPIO32
scan: true
frequency: 400kHz

es8311:
address: 0x18

es7210:
address: 0x40

output:

platform: gpio
id: pa_ctrl
pin: GPIO12
i2s_audio:

id: codec
i2s_lrclk_pin: GPIO22
i2s_bclk_pin: GPIO25
i2s_mclk_pin: GPIO0
id: mic_adc
i2s_lrclk_pin: GPIO26
i2s_bclk_pin: GPIO27
i2s_mclk_pin: GPIO0
speaker:

platform: i2s_audio
id: external_speaker
dac_type: external
i2s_audio_id: codec
i2s_dout_pin: GPIO13
mode: mono
microphone:

platform: i2s_audio
id: external_mic
adc_type: external
i2s_audio_id: mic_adc
i2s_din_pin: GPIO36
pdm: false
voice_assistant:
id: voice_asst
microphone: external_mic
speaker: external_speaker
noise_suppression_level: 2
auto_gain: 15dBFS
volume_multiplier: 0.5
use_wake_word: false
on_listening:
- light.turn_on:
id: led_ring
blue: 100%
red: 0%
green: 0%
brightness: 100%
effect: wakeword
on_tts_start:
- light.turn_on:
id: led_ring
blue: 0%
red: 0%
green: 100%
brightness: 50%
effect: pulse
on_end:
- delay: 100ms
- wait_until:
not:
speaker.is_playing:
- script.execute: reset_led
on_error:
- light.turn_on:
id: led_ring
blue: 0%
red: 100%
green: 0%
brightness: 100%
effect: none
- delay: 1s
- script.execute: reset_led
- script.wait: reset_led
- lambda: |-
if (code == “wake-provider-missing” || code == “wake-engine-missing”) {
id(use_wake_word).turn_off();
}

script:

id: reset_led
then:
if:
condition:
switch.is_on: use_wake_word
then:
- light.turn_on:
id: led_ring
blue: 30%
red: 0%
green: 0%
brightness: 25%
effect: none
else:
- light.turn_off: led_ring
switch:

platform: template
name: Use wake word
id: use_wake_word
optimistic: true
restore_mode: RESTORE_DEFAULT_ON
entity_category: config
on_turn_on:
lambda: id(voice_asst).set_use_wake_word(true);
if:
condition:
not:
- voice_assistant.is_running
then:
- voice_assistant.start_continuous
script.execute: reset_led
on_turn_off:
voice_assistant.stop
script.execute: reset_led
Restart option
platform: restart
name: “${device_name} Restart”
light:

platform: esp32_rmt_led_strip
id: led_ring
name: “${device_name} Light”
pin: GPIO33
num_leds: 12
rmt_channel: 0
rgb_order: GRB
chipset: ws2812
default_transition_length: 0s
effects:
pulse:
name: “Pulse”
transition_length: 0.5s
update_interval: 0.5s
addressable_twinkle:
name: “Working”
twinkle_probability: 5%
progress_interval: 4ms
addressable_color_wipe:
name: “Wakeword”
colors:
- red: 28%
green: 100%
blue: 90%
num_leds: 12
add_led_interval: 40ms
reverse: false
binary_sensor:

platform: template
name: “${device_name} Volume Up”
id: btn_volume_up
platform: template
name: “${device_name} Volume Down”
id: btn_volume_down
platform: template
name: “${device_name} Set”
id: btn_set
platform: template
name: “${device_name} Play”
id: btn_play
platform: template
name: “${device_name} Mode”
id: btn_mode
platform: template
name: “${device_name} Record”
id: btn_record
on_press:
output.turn_on: pa_ctrl
voice_assistant.start:
light.turn_on:
id: led_ring
brightness: 100%
effect: “Wakeword”
on_release:
voice_assistant.stop:
output.turn_off: pa_ctrl
light.turn_off:
id: led_ring
Status de connection
platform: status
name: “${device_name} Status”
sensor:

id: button_adc
platform: adc
internal: true
pin: 39
attenuation: 11db
update_interval: 15ms
filters:
median:
window_size: 5
send_every: 5
send_first_at: 1
delta: 0.1
on_value_range:
below: 0.55
then:
binary_sensor.template.publish:
id: btn_volume_up
state: ON
above: 0.65
below: 0.92
then:
binary_sensor.template.publish:
id: btn_volume_down
state: ON
above: 1.02
below: 1.33
then:
binary_sensor.template.publish:
id: btn_set
state: ON
above: 1.43
below: 1.77
then:
binary_sensor.template.publish:
id: btn_play
state: ON
above: 1.87
below: 2.15
then:
binary_sensor.template.publish:
id: btn_mode
state: ON
above: 2.25
below: 2.56
then:
binary_sensor.template.publish:
id: btn_record
state: ON
above: 2.8
then:
binary_sensor.template.publish:
id: btn_volume_up
state: OFF
binary_sensor.template.publish:
id: btn_volume_down
state: OFF
binary_sensor.template.publish:
id: btn_set
state: OFF
binary_sensor.template.publish:
id: btn_play
state: OFF
binary_sensor.template.publish:
id: btn_mode
state: OFF
binary_sensor.template.publish:
id: btn_record
state: OFF
Wifi signal
platform: wifi_signal
name: “${device_name} WiFi Signal”
update_interval: 60s`

Can you help to adapt this for a full work rhaaspy sattellite.

Thanks

Romkabouter, 2024-01-19:
Sorry, this repo is not for ESPHome

closed 2024-01-19 by Romkabouter

orrionis, 2024-01-19:
I know. 
I want to transform this esphome code for use with rhasspy without esphome

I post this code, for hello if needed

⁣Télécharger BlueMail pour Android ​

Romkabouter, 2024-01-19:
If you want to use this repo with Rhasspy, you must create a new device and specify the correct GPIO values matching your device. In the readme there is a section about adding new devices.
Hope it helps.

But if your device works with Esphome, I would personally go with that :)

orrionis, 2024-01-19:
Humm ok . This device inbesphome can work whith rhasppy ? Via home assistant ?

⁣Télécharger BlueMail pour Android ​

Romkabouter, 2024-01-19:
I don't know, you posted the esphome code so I was assuming yes  :)